yc9559/uperf #12: Redmi Note 5 进入系统几分钟后开始卡死 重启

ahzhi, 2021-01-31:
SOC=SDM636 (EAS 内核）
Android = 11
Magisk Version= 21.4

Hi 大佬 我手机在进入系统后就开始卡死 完全不能操作 过了几分钟后手机就会自动重启 然后就会不断的循环这个问题
只能关掉uperf来解决问题
我复现这些问题和抓了log
希望能帮助到您

谢谢你
[dmesg.log](https://github.com/yc9559/uperf/files/5900085/dmesg.log)
[logcat.log](https://github.com/yc9559/uperf/files/5900086/logcat.log)

yc9559, 2021-02-27:
libperfmgr抛出`Invalid config: /vendor/etc/powerhint.json`已修复，问题在于空配置内未定义键值引用。

closed 2021-02-27 by yc9559

ahzhi, 2021-02-28:
不好意思大佬，打扰了。
在我用 DEV 21.01.28 版本的时候 还是发生了手机卡机重启
我通过复现操作与抓取log

希望这些log 能帮助到你

[dmesg.log](https://github.com/yc9559/uperf/files/6057321/dmesg.log)

[logcat.log](https://github.com/yc9559/uperf/files/6057320/logcat.log)

yc9559, 2021-03-03:
谢谢，已修复，问题在于`libperfmgr: Action[11]'s NodeIndex is duplicated with another Action`

ahzhi, 2021-03-03:
谢谢大佬