yc9559/uperf #13: 猎户座9820会在开机几分钟之内睡死

emptybot08, 2021-02-01:
开机后，过将近三分钟之后会睡死无法正常点亮屏幕，但是如果连接电脑或是充电则不会睡死，无线adb连接后，睡死时依然有效并在抓取logcat一段时间后手机自动重启，以下是我使用无线adb调试所抓取的日志文件，希望可以起到作用，机型三星galaxy s10 韩版，cpu猎户座9820，系统为one ui2.5，android10
[logcat.log](https://github.com/yc9559/uperf/files/5901813/logcat.log)
[dmesg.log](https://github.com/yc9559/uperf/files/5901814/dmesg.log)

Saumer7, 2021-02-01:
I think I have the same problem. galaxy s10 exynos 9820 korean version, RR rom (LieangeOS based) installed.
enable uperf and screen off, wait some minute. then phone 'wont wake up' or 'get automatic reboot and stuck in bootloop(Manual reboot need to escape from bootloop)'
the problem was exist on uperf v1 too.
if disable cpuFreqMin (open cfg_modify.json and change enable value true to false), the problem disappears.

yc9559, 2021-02-15:
@Saumer7 Thank you, fixed in DEV 21.02.14.

closed 2021-02-15 by yc9559