yc9559/uperf #16: 能否通过使用 sepolicy 帮助 sfanalysis 完成注入而不是直接把 selinux 设为宽容

rhjdvsgsgks, 2021-02-03:
rt，最近见很多人在讨论这个，听说 selinux 设为宽容不安全，找了一圈最后发现是 uperf 关掉的，有人说可以用 sepolicy 代替关 selinux ，不知能否实现

zhangsiyu0, 2021-02-09:
uperf没关selinux，你再检测下其它模块，或者你用的系统本身就是关的。

rhjdvsgsgks, 2021-02-09:
> uperf没关selinux，你再检测下其它模块，或者你用的系统本身就是关的。

https://github.com/yc9559/uperf/blob/4070883fb02cbd8f0f807241b04742f7ea2c3105/magisk/script/start_sfanalysis.sh#L46

我是在已经确定了的情况下才发的 issue 的，请问您是如何得出“uperf没关selinux”这个结论的？

zhangsiyu0, 2021-02-09:
> > uperf没关selinux，你再检测下其它模块，或者你用的系统本身就是关的。
> 
> https://github.com/yc9559/uperf/blob/4070883fb02cbd8f0f807241b04742f7ea2c3105/magisk/script/start_sfanalysis.sh#L46
> 
> 我是在已经确定了的情况下才发的 issue 的，请问您是如何得出“uperf没关selinux”这个结论的？

额，抱歉没怎么看代码，因为我用了很久都没影响selinux，刚看了下确实有改变selinux的可能。

bwcxwlx, 2021-02-11:
Readme中SfAnalysis写有
[image]
且setenforce 0命令是在SELinux打开/强制模式下无法正常注入使用时，才会使用。注释写得很清楚了。
[image]
可能是你的系统在SELinux打开/强制时无法使用，所以最好不使用或者更准确地反馈，而不是简单地认为Uperf关闭/宽容SELinux。
在我的X7Pro上，可以在保持着SELinux强制模式使用Uperf模块。

rhjdvsgsgks, 2021-02-11:
> Readme中SfAnalysis写有
> [image]
> 且setenforce 0命令是在SELinux打开/强制模式下无法正常注入使用时，才会使用。注释写得很清楚了。
> [image]
> 可能是你的系统在SELinux打开/强制时无法使用，所以最好不使用或者更准确地反馈，而不是简单地认为Uperf关闭/宽容SELinux。
> 在我的X7Pro上，可以在保持着SELinux强制模式使用Uperf模块。

这些我都是知道的，只是我觉得在发 issue 的时候没必要把这些再提一遍  
而且我认为即使是在某些条件下才会有某种行为，这也并不能否认“有某种行为”的事实

bwcxwlx, 2021-02-11:
> > Readme中SfAnalysis写有
> > [image]
> > 且setenforce 0命令是在SELinux打开/强制模式下无法正常注入使用时，才会使用。注释写得很清楚了。
> > [image]
> > 可能是你的系统在SELinux打开/强制时无法使用，所以最好不使用或者更准确地反馈，而不是简单地认为Uperf关闭/宽容SELinux。
> > 在我的X7Pro上，可以在保持着SELinux强制模式使用Uperf模块。
> 
> 这些我都是知道的，只是我觉得在发 issue 的时候没必要把这些再提一遍
> 而且我认为即使是在某些条件下才会有某种行为，这也并不能否认“有某种行为”的事实

好的 很抱歉，也希望作者能看到，进而共同找到问题所在并做出可能的调整。
但是我还是认为写清楚比较合适，能减少不必要的沟通也有助于问题解决。
另外说句题外话，我个人不是很认可你说的“某种条件”（我在受到生命威胁时做出反击≠我伤害他人）。当然也没有在这里讨论这个的必要。

rhjdvsgsgks, 2021-02-11:
> 好的 很抱歉，也希望作者能看到，进而共同找到问题所在并做出可能的调整。
> 但是我还是认为写清楚比较合适，能减少不必要的沟通也有助于问题解决。
> 另外说句题外话，我个人不是很认可你说的“某种条件”（我在受到生命威胁时做出反击≠我伤害他人）。当然也没有在这里讨论这个的必要。

没有写出来是因为只有一种条件会触发关闭 selinux ，相信作者应该比我们都清楚
“某种条件”那句话的意思是“在尝试注入失败后会关闭 selinux 也是会关闭 selinux“，你举的例子也一样，”我受到威胁做出反击确实不等于我(主动)伤害他人，但哪怕是受到威胁做出反击，他人受到了来自我的伤害也依然是个存在的客观事实“

bwcxwlx, 2021-02-11:
> > 好的 很抱歉，也希望作者能看到，进而共同找到问题所在并做出可能的调整。
> > 但是我还是认为写清楚比较合适，能减少不必要的沟通也有助于问题解决。
> > 另外说句题外话，我个人不是很认可你说的“某种条件”（我在受到生命威胁时做出反击≠我伤害他人）。当然也没有在这里讨论这个的必要。
> 
> 没有写出来是因为只有一种条件会触发关闭 selinux ，相信作者应该比我们都清楚
> “某种条件”那句话的意思是“在尝试注入失败后会关闭 selinux 也是会关闭 selinux“，你举的例子也一样，”我受到威胁做出反击确实不等于我(主动)伤害他人，但哪怕是受到威胁做出反击，他人受到了来自我的伤害也依然是个客观存在的事实“

嗯 我的锅
感谢，新年快乐

yc9559, 2021-02-27:
@rhjdvsgsgks DEV21.02.28将会移除隐式SELinux操作，如果出现注入失败，请安装关闭SELinux的模块并确保在注入之前执行关闭SELinux。

closed 2021-02-27 by yc9559