yc9559/uperf #30: 请不要在sdcard根目录创建文件夹

Gokou-Ruri, 2021-04-10:
各种应用都在sdcard创建目录的话真的非常乱，希望不要采用这种策略
个人建议像magisk本体那样在/data中创建目录，或者像riru等模块那样，在/data/local中创建对应的目录。
或者在/sdcard/Document目录中创建也可以。

而且qti-mem模块依旧还是在/sdcard/Android目录中创建文件，但uperf又在sdcard根目录创建目录和文件，操作上也不统一（查看log和修改参数等）。
再者，从Android 11开始，/sdcard更换为了fuse文件系统，小文件随机读写延迟比sdcardfs提高了数倍，在/sdcard中读写log文件也并不合适，更建议开发者在ext4/f2fs物理文件系统中读写文件。

AngelSherry, 2021-04-19:
我個人也認為開發者將以前的/sdcard/Android/yc改為/sdcard/yc這一行為很不妥，也十分不能理解為什麼開發者會做出這樣的改動。

在Android 11上Google限制的是/sdcard/Android/data的讀取，而又不是/sdcard/Android的讀取，所以一般的檔案管理器是可以直接修改/sdcard/Android/yc的內容的。

至於說擔心用戶找不到Android檔案在哪則更沒必要，會刷入yc調度的用戶會找不到Android檔案我是根本不相信的。

現在yc的配置檔存在/sdcard下感覺和很多需要「存儲重定向」的毒瘤APP沒什麼區別，本身/sdcard下就不應該存放日誌以及配置檔來著的說。

railjty, 2022-01-06:
这种随时可能会自己修改的还是放sdcard下面好点，而且就多个yc/uperf，作者的Dfps也在yc/dfps下面，同一个作者用同一个文件夹存也不是不行。
而且安卓11有各种办法关掉fuse这个问题根源，何必抱着这个低性能的东西不放而去适应它？
还有在非模块文件夹下面放东西就是模块删不干净必须格式化data分区的原因，riru那个做法其实更离谱，还有在Android下面放配置的也是一样。
如果非要整洁的/sdcard，请删除所有中国App（重定向已经不能阻止这些App在/sdcard/Document随意创建文件夹用于追踪）并格机（激进的可以删掉所有MinSDK<30的），然后使用CuToolbox（不需要配置文件，但是MinSDK=28）

Gokou-Ruri, 2022-01-07:
> 这种随时可能会自己修改的还是放sdcard下面好点，而且就多个yc/uperf，作者的Dfps也在yc/dfps下面，同一个作者用同一个文件夹存也不是不行。 而且安卓11有各种办法关掉fuse这个问题根源，何必抱着这个低性能的东西不放而去适应它？ 还有在非模块文件夹下面放东西就是模块删不干净必须格式化data分区的原因，riru那个做法其实更离谱，还有在Android下面放配置的也是一样。 [...]

1. “一个作者一个文件夹”，请问你的手机只安装一个uperf，不安装其它任何程序、模块吗？举个比较粗俗的例子：每个人都各自固定在公共场合的某一个地方排泄就不是随地大小便了吗？每个程序都抱着“反正我的文件都在同一个文件夹里”的思想去设计，每个软件都这样，sdcard里才会有那么多垃圾文件夹。Android虽然没有彻底禁止往/sdcard目录读写，但这不代表程序猿就应该滥用/sdcard根目录，而且Android已经为开发者准备了诸多可供读写的数据目录，何况uperf还具备root权限。就像随地大小便虽然不违法，但明显和道德观念相悖。
2. 先不说Android 11+从fuse切换到sdcardfs会导致一些问题（例如部分APP因为遵循了新的规范无法在Android 11+读取sdcardfs，以及切换到sdcardfs会导致`/storage/emulated/0/Android/data`下的APP数据目录失去隐私性和安全性等），出厂Android 11的设备内核里根本没有sdcardfs驱动，请问你要怎么切换到sdcardfs？
3. 第一次听说模块删不干净必须格式化data这种离谱说法，模块在哪里创建文件，删除对应的文件不就好了？Linux系统本来就是这样的使用方式。
4. “如果非要整洁的/sdcard，请删除所有中国App”，这又是什么迷惑发言？“不随地大小便不是中国人”？我的手机里这么多国内APP，也没见哪里不干净。包括媒体目录里也也不存在跟踪文件，你这些言论从何得出？我向开发者提建议不要在/sdcard下面胡乱创建目录，你不赞同是你的自由，但你在这儿公然反对是出于什么心态？在/sdcard释放乱七八糟的文件夹和追踪文件的只是部分国产流氓APP而已，你是出于什么思路将流氓软件的行为上升到所有的“中国APP”？难道你的意思是那些遵规守矩干干净净的APP开发者都不是中国人？或者中国人就不应该有素质有道德？

railjty, 2022-01-07:
> > 这种随时可能会自己修改的还是放sdcard下面好点，而且就多个yc/uperf，作者的Dfps也在yc/dfps下面，同一个作者用同一个文件夹存也不是不行。 而且安卓11有各种办法关掉fuse这个问题根源，何必抱着这个低性能的东西不放而去适应它？ 还有在非模块文件夹下面放东西就是模块删不干净必须格式化data分区的原因，riru那个做法其实更离谱，还有在Android下面放配置的也是一样。 如果非要整洁的/sdcard，请删除所有中国App（重定向已经不能阻止这些App在/sdcard/Document随意创建文件夹用于追踪）并格机（激进的可以删掉所有MinSDK<30的），然后使用CuToolbox（不需要配置文件，但是MinSDK=28）
> 
> 1. “这种随时可能会自己修改的还是放sdcard下面好点”，举个例子，例如放在`/sdcard/Android`或者`/sdcard/Document`也仅仅只是多一个子目录层级而已，哪怕是放在/data分区的某个文件夹中，也可以添加书签一键直达，比起放在`/sdcard/yc`且/sdcard中有一大堆其它APP产生的垃圾文件夹还得滑动多次才能进入`yc`目录的情况相比不是更快吗？
> 2. “一个作者一个文件夹”，请问你的手机只安装一个uperf，不安装其它任何程序、模块吗？举个比较粗俗的例子：每个人都各自固定在公共场合指定厕所以外的某一个地方排泄就不是随地大小便了吗？每个程序都抱着“反正我的文件都在同一个文件夹里”的思想去设计，每个软件都这样，sdcard里才会有那么多垃圾文件夹。Android虽然没有彻底禁止往/sdcard目录读写，但这不代表程序猿就应该滥用/sdcard根目录，而且Android已经为开发者准备了诸多可供读写的数据目录，何况uperf还具备root权限。就像随地大小便虽然不违法，但明显和道德观念相悖。
> 3. 先不说Android 11+从fuse切换到sdcardfs会导致一些问题（例如部分APP因为遵循了新的规范无法在Android 11+读取sdcardfs，以及切换到sdcardfs会导致`/storage/emulated/0/Android/data`下的APP数据目录失去隐私性和安全性等），出厂Android 11的设备内核里根本没有sdcardfs驱动，请问你要怎么切换到sdcardfs？
> 4. 第一次听说模块删不干净必须格式化data这种离谱说法，模块在哪里创建文件，删除对应的文件不就好了？Linux系统本来就是这样的使用方式。
> 5. “如果非要整洁的/sdcard，请删除所有中国App”，这又是什么迷惑发言？“不随地大小便不是中国人”？我的手机里这么多国内APP，也没见哪里不干净。包括媒体目录里也也不存在跟踪文件，你这些言论从何得出？我向开发者提建议不要在/sdcard下面胡乱创建目录，你不赞同是你的自由，但你在这儿公然反对是出于什么心态？在/sdcard释放乱七八糟的文件夹和追踪文件的只是部分国产流氓APP而已，你是出于什么思路将流氓软件的行为上升到所有的“中国APP”？难道你的意思是那些遵规守矩干干净净的APP开发者都不是中国人？或者中国人就不应该有素质有道德？

我是反对APP追踪的，但是现在谷歌摆烂，允许app读取文件夹名字，所以保存用户ID和原来几乎一样没限制，而且可以预见BAT等需要追踪的APP和广告SDK（不是反对所有国内APP。。。是我当时确实快被这些东西搞疯了）都将使用这种方法。
不是说中国人没有道德，而是只要你开始装国内版，就不可能避免他们创建各种文件夹，目前只能寄希望于良心发现，所以说了气话。
还有，模块删不干净是真的有的，不能保证每个作者都那么完美。
同时因为模块的自由性，故最终建议你自己解包修改这些路径配置

Gokou-Ruri, 2022-01-07:
> 

可是很遗憾的告诉你，我用sed改完模块内所有纯文本文件的路径且grep查找不到任何含`/sdcard/yc`的内容以后，在模块产生的log文件内依旧会出现`/sdcard/yc`的字段（找不到`cur_powermode`文件），这就是为什么我来向开发者提建议的原因，毕竟这个模块它并不开源，开源的只是SoC的调度配置文件。

Gokou-Ruri, 2022-01-07:
> 我是反对APP追踪的，但是现在谷歌摆烂，允许app读取文件夹名字，所以保存用户ID和原来几乎一样没限制，而且可以预见BAT等需要追踪的APP和广告SDK（不是反对所有国内APP。。。是我当时确实快被这些东西搞疯了）都将使用这种方法。 不是说中国人没有道德，而是只要你开始装国内版，就不可能避免他们创建各种文件夹，目前只能寄希望于良心发现，所以说了气话。 还有，模块删不干净是真的有的，不能保证每个作者都那么完美。 同时因为模块的自由性，故最终建议你自己解包修改这些路径配置

1. 你说的Android 11+流氓软件做用户跟踪和过去一样这点我知道，我相信哪怕不是开发者，只要稍微对文件系统有点常识都会明白这一点，但关于这一点，我觉得我作为普通用户是无能为力的，Android不管变化成什么样我都只能用它，因为目前市面上只有两款主流的手机操作系统。
2. 虽说“有什么用什么”，但这些问题并非没有解决办法，总会有优秀的开发者开发出实用的工具来解决谷歌这个流氓制造的问题，例如`存储空间隔离`。
3. “模块删不干净是真的有的，不能保证每个作者都那么完美。”，前文说过，我提建议就是想让开发者尽可能遵守规范，让这些程序尽量接近完美，不希望开发者在不恰当的位置随便写入文件。而你碰到的删不干净的情况只可能是模块的开发者不遵守规范导致的——例如胡乱修改系统文件却没有进行备份，而riru和riru衍生的那些模块并不存在这种问题，所以这和我对开发者的建议并不冲突，我觉得你不应该来反对我向uperf开发者提建议，何况你也应该能看到不止我一个人被这个问题困扰着。
4. magisk模块并非只能挂载文件，它也能删改文件，毕竟模块安装后就具有了root权限，理论上来说模块想干什么都可以，如果模块永久性修改了某个系统文件，当然就会出现无法还原（删干净模块造成的影响）的情况。所以才需要开发者有意识地去规范模块的行为，而不是每个开发者都想怎么样就怎么样，比如都在/sdcard下面创建自己专用的子目录。

railjty, 2022-01-07:
> > > > 这种随时可能会自己修改的还是放sdcard下面好点，而且就多个yc/uperf，作者的Dfps也在yc/dfps下面，同一个作者用同一个文件夹存也不是不行。 而且安卓11有各种办法关掉fuse这个问题根源，何必抱着这个低性能的东西不放而去适应它？ 还有在非模块文件夹下面放东西就是模块删不干净必须格式化data分区的原因，riru那个做法其实更离谱，还有在Android下面放配置的也是一样。 如果非要整洁的/sdcard，请删除所有中国App（重定向已经不能阻止这些App在/sdcard/Document随意创建文件夹用于追踪）并格机（激进的可以删掉所有MinSDK<30的），然后使用CuToolbox（不需要配置文件，但是MinSDK=28）
> > > 
> > > 
> > > 
> > > 1. “这种随时可能会自己修改的还是放sdcard下面好点”，举个例子，例如放在`/sdcard/Android`或者`/sdcard/Document`也仅仅只是多一个子目录层级而已，哪怕是放在/data分区的某个文件夹中，也可以添加书签一键直达，比起放在`/sdcard/yc`且/sdcard中有一大堆其它APP产生的垃圾文件夹还得滑动多次才能进入`yc`目录的情况相比不是更快吗？
> > > 2. “一个作者一个文件夹”，请问你的手机只安装一个uperf，不安装其它任何程序、模块吗？举个比较粗俗的例子：每个人都各自固定在公共场合指定厕所以外的某一个地方排泄就不是随地大小便了吗？每个程序都抱着“反正我的文件都在同一个文件夹里”的思想去设计，每个软件都这样，sdcard里才会有那么多垃圾文件夹。Android虽然没有彻底禁止往/sdcard目录读写，但这不代表程序猿就应该滥用/sdcard根目录，而且Android已经为开发者准备了诸多可供读写的数据目录，何况uperf还具备root权限。就像随地大小便虽然不违法，但明显和道德观念相悖。
> > > 3. 先不说Android 11+从fuse切换到sdcardfs会导致一些问题（例如部分APP因为遵循了新的规范无法在Android 11+读取sdcardfs，以及切换到sdcardfs会导致`/storage/emulated/0/Android/data`下的APP数据目录失去隐私性和安全性等），出厂Android 11的设备内核里根本没有sdcardfs驱动，请问你要怎么切换到sdcardfs？
> > > 4. 第一次听说模块删不干净必须格式化data这种离谱说法，模块在哪里创建文件，删除对应的文件不就好了？Linux系统本来就是这样的使用方式。
> > > 5. “如果非要整洁的/sdcard，请删除所有中国App”，这又是什么迷惑发言？“不随地大小便不是中国人”？我的手机里这么多国内APP，也没见哪里不干净。包括媒体目录里也也不存在跟踪文件，你这些言论从何得出？我向开发者提建议不要在/sdcard下面胡乱创建目录，你不赞同是你的自由，但你在这儿公然反对是出于什么心态？在/sdcard释放乱七八糟的文件夹和追踪文件的只是部分国产流氓APP而已，你是出于什么思路将流氓软件的行为上升到所有的“中国APP”？难道你的意思是那些遵规守矩干干净净的APP开发者都不是中国人？或者中国人就不应该有素质有道德？
> > 
> > 
> > 那好，我可以告诉你修改文件夹路径的地方：scripts/pathinfo.sh，自己修改吧，这都能改的，启动时候读这个脚本里面写的目录
> 
> 可是很遗憾的告诉你，我用sed改完模块内所有纯文本文件的路径且grep查找不到任何含`/sdcard/yc`的内容以后，在模块产生的log文件内依旧会出现`/sdcard/yc`的字段（找不到`cur_powermode`文件），这就是为什么我来向开发者提建议的原因，毕竟这个模块它并不开源，开源的只是SoC的调度配置文件。

那个cur_mode在/data/media/0/yc/uperf/cfg_uperf.json（也就是调度的配置文件）里面，也要对应改，我能帮你的就这么多了
我也挺希望这些模块作者好好规范的（还有在/sdcard/Android下面创建配置的），无奈Magisk Alpha没有解决部分情况掉模块问题（掉过riru,zy版本lsposed），不太敢直接修改到模块目录下面

railjty, 2022-01-07:
> 可是很遗憾的告诉你，我用sed改完模块内所有纯文本文件的路径且grep查找不到任何含`/sdcard/yc`的内容以后，在模块产生的log文件内依旧会出现`/sdcard/yc`的字段（找不到`cur_powermode`文件），这就是为什么我来向开发者提建议的原因，毕竟这个模块它并不开源，开源的只是SoC的调度配置文件。

如果你愿意，可以去看看我在gitee上魔改的yc调度最新的commit（着重关注pathinfo.sh,mtd1200.json)，里面示范性修改/sdcard为/data/media/0

Gokou-Ruri, 2022-01-07:
> 那个cur_mode在/data/media/0/yc/uperf/cfg_uperf.json（也就是调度的配置文件）里面，也要对应改，我能帮你的就这么多了 我也挺希望这些模块作者好好规范的（还有在/sdcard/Android下面创建配置的），无奈Magisk Alpha没有解决部分情况掉模块问题（掉过riru,zy版本lsposed），不太敢直接修改到模块目录下面

是我的表达方式有误还是什么？但我觉得我应该没有词不达意的地方吧？
[quote]我用sed改完模块内所有纯文本文件的路径且grep查找不到任何含/sdcard/yc的内容以后，在模块产生的log文件内依旧会出现/sdcard/yc的字段（找不到cur_powermode文件），这就是为什么我来向开发者提建议的原因，毕竟这个模块它并不开源，开源的只是SoC的调度配置文件。[/quote]
你想过为什么要改模块安装以后生成的配置文件才能解决模块配置文件路径的问题吗？你在打算指点我之前，能否先仔细看看我的回复内容呢？（事实上我真不想与你继续这个话题，你我都明白GitHub并非聊天场所，且此话题继续下去也没有意义，但我不想close这个问题，所以真心希望你不要再继续下去，望克制。）
再者，我姑且退一步说，哪怕你面对的是一个什么都不懂的小白，对方给开发者提建议不要在/sdcard创建垃圾文件，你却一而再再而三阻扰以及试图“教导”对方用并不算完美的解决方案来曲线解决问题，有必要吗？这种事让开发者自己修改自己的闭源代码一劳永逸不是更好吗？sdcard里少一个垃圾目录不是更好吗？在GitHub提建议的人并不算多，但此问题点赞数却已过30，显然被垃圾目录困扰的并不止我一人。
话题一直被你越扯越远。“无奈Magisk Alpha没有解决部分情况掉模块问题（掉过riru,zy版本lsposed），不太敢直接修改到模块目录下面”我可从来没说过建议开发者一定要把config和log相关文件放到模块所在目录里，而且magisk alpha自身的稳定性也和此issue毫无关系。

railjty, 2022-01-07:
OK,这个就当建议提给yc9559

yc9559, 2022-04-08:
已将目录改回`/sdcard/Android/yc`，此前经常有人找不到藏在Android目录的路径

closed 2022-04-08 by yc9559