yc9559/uperf #37: It's changing my minimal frequency for both big and little core

duepii, 2021-05-08:
Had this issue with your module

I can't modify those even from kernel side, it locks minimal frequency for both cores to an higher value than usual, in my case 979 mhz for big for example when my usual min freq is 300 mhz and I can't change that in any way.

I fixed removing some lines from your module by myself, I think your module should not edit min and max freq in any way actually

eebssk1, 2021-05-14:
There's no way to alter the target frequency without chaging the min/max frequency range.
Anyway you can modify the config to suit your need.

duepii, 2021-05-14:
Point is your module should not touch frequencies imho

eebssk1, 2021-05-15:
> 
> 
> Point is your module should not touch frequencies imho

Well, the module simply take over the QTI perfd 's job which do the same but comes with your device by default.
Which means the frequency is controlled by system 's default daemon but now managed by this module.
If you want to controlled the frequency yourself or by governer itself then you need to disable this module and stop the system performance daemon.

yc9559, 2021-07-22:
@duepii Well, you can disable CPU frequency range override by setting these knobs to false:
```json
        "knobs": [
            {
                "name": "cpuFreqMax",
                "path": "/sys/devices/system/cpu/cpu%d/cpufreq/scaling_max_freq",
                "type": "cpufreq",
                "enable": false
            },
            {
                "name": "cpuFreqMin",
                "path": "/sys/devices/system/cpu/cpu%d/cpufreq/scaling_min_freq",
                "type": "cpufreq",
                "enable": false
            },
]
```

closed 2021-07-22 by yc9559