yc9559/uperf #45: Hot Temperature Detected !!!

Virgosan, 2021-07-05:
After i install the module the phone feeling hot 49/50 celcius after 10 minutes and im trying to playing game...i found the issue it because of the system/vendor/etc/perf/common&target make my phone hot after remove it and reflash again my temperature become normal

ps/: My phone is Surya>Snapdragon 732g

yc9559, 2021-07-22:
Execute `su -c top` to find which process is consuming CPU.

Virgosan, 2021-07-26:
[image]

railjty, 2022-01-06:
> [image]

No process name? Please use adb shell "su -c top"and see which process is consuming CPU

closed 2022-04-21 by yc9559