yc9559/uperf #59: 8.15版本与miui开发版冲突

sino1641, 2021-08-22:
k30s 开发版8.18 magisk 23001
刷入重启进入桌面后软重启😭
回退7.25版后正常使用

1. 酷安上mi10等高通系手机均有类似情况，
<details>
<summary>附调试日志：</summary>
```
[00:43:49][I] Uperf v2 (21.08.15), by Matt Yang (yccy@outlook.com)
[00:43:49][I] Daemon: Start
[00:43:49][I] Daemon: New uperf(pid=22363) is loading config file
[00:43:49][I] Uperf: Initializing
[00:43:49][I] CfgMgr: Using [sdm865/sdm865+/sdm870[21.08.15]] by [yc@coolapk]
[00:43:49][I] CfgMgr: Verbose log enabled
[00:43:49][I] CfgMgr: Bind None -> normal
[00:43:49][I] CfgMgr: Bind Active -> active
[00:43:49][I] CfgMgr: Bind Touch -> interaction
[00:43:49][I] CfgMgr: Bind Pressed -> pressed
[00:43:49][I] CfgMgr: Bind Tap -> interaction
[00:43:49][I] CfgMgr: Bind Swipe -> interaction
[00:43:49][I] CfgMgr: Bind HeavyLoad -> heavyLoad
[00:43:49][I] CfgMgr: Bind SfLag -> sfLag
[00:43:49][I] CfgMgr: Bind SfBoost -> sfBoost
[00:43:49][I] CfgMgr: Bind SsAnim -> heavyLoad
[00:43:49][I] CfgMgr: Bind Wakeup -> heavyLoad
[00:43:49][V] CfgMgr: Start parsing powermode "balance"
[00:43:49][V] CfgMgr: Ignored knobs in action root/powermodes/balance/actions/sfLag:
[00:43:49][V] CfgMgr: schedBoost topBoost heavyBoost llccBwMax llccBwMin ddrBwMax ddrBwMin l3LatBig ddrLatBig dispFeature 
[00:43:49][V] CfgMgr: Ignored knobs in action root/powermodes/balance/actions/sfBoost:
[00:43:49][V] CfgMgr: schedBoost l3LatBig ddrLatBig dispFeature 
[00:43:49][V] CfgMgr: Start parsing powermode "powersave"
[00:43:49][V] CfgMgr: Ignored knobs in action root/powermodes/powersave/actions/sfLag:
[00:43:49][V] CfgMgr: schedBoost topBoost heavyBoost llccBwMax llccBwMin ddrBwMax ddrBwMin l3LatBig ddrLatBig dispFeature 
[00:43:49][V] CfgMgr: Ignored knobs in action root/powermodes/powersave/actions/sfBoost:
[00:43:49][V] CfgMgr: schedBoost l3LatBig ddrLatBig dispFeature 
[00:43:49][V] CfgMgr: Start parsing powermode "performance"
[00:43:49][V] CfgMgr: Ignored knobs in action root/powermodes/performance/actions/sfLag:
[00:43:49][V] CfgMgr: schedBoost topBoost heavyBoost llccBwMax llccBwMin ddrBwMax ddrBwMin l3LatBig ddrLatBig dispFeature 
[00:43:49][V] CfgMgr: Ignored knobs in action root/powermodes/performance/actions/sfBoost:
[00:43:49][V] CfgMgr: schedBoost l3LatBig ddrLatBig dispFeature 
[00:43:49][I] CfgMgr: Read default powermode from /sdcard/yc/uperf/cur_powermode
[00:43:49][V] CfgMgr: Bind None -> balance.normal
[00:43:49][V] CfgMgr: Bind Active -> balance.active
[00:43:49][V] CfgMgr: Bind Touch -> balance.interaction
[00:43:49][V] CfgMgr: Bind Pressed -> balance.pressed
[00:43:49][V] CfgMgr: Bind Tap -> balance.interaction
[00:43:49][V] CfgMgr: Bind Swipe -> balance.interaction
[00:43:49][V] CfgMgr: Bind HeavyLoad -> balance.heavyLoad
[00:43:49][V] CfgMgr: Bind SfLag -> balance.sfLag
[00:43:49][V] CfgMgr: Bind SfBoost -> balance.sfBoost
[00:43:49][V] CfgMgr: Bind SsAnim -> balance.heavyLoad
[00:43:49][V] CfgMgr: Bind Wakeup -> balance.heavyLoad
[00:43:49][I] CfgMgr: Powermode "(null)" -> "balance"
[00:43:49][V] Watcher: Add watching path=/sdcard/yc/uperf/cur_powermode mask=8
[00:43:49][V] Watcher: Add watching path=/dev/cpuset/top-app/tasks mask=2
[00:43:49][V] Watcher: Add watching path=/dev/cpuset/foreground/tasks mask=2
[00:43:49][V] Watcher: Add watching path=/dev/cpuset/background/tasks mask=2
[00:43:49][V] Watcher: Add watching path=/dev/cpuset/restricted/tasks mask=2
[00:43:49][I] InputReader: Enabled by config file
[00:43:49][V] InputReader: opening /dev/input/event8
[00:43:49][I] InputReader: btn:"uinput-fpc"
[00:43:49][V] InputReader: opening /dev/input/event7
[00:43:49][V] InputReader: opening /dev/input/event6
[00:43:49][I] InputReader: btn:"kona-mtp-snd-card Button Jack"
[00:43:49][V] InputReader: opening /dev/input/event5
[00:43:49][V] InputReader: opening /dev/input/event4
[00:43:49][I] InputReader: touchpanel:"NVTCapacitiveTouchScreen" x:0~1079 y:0~2399
[00:43:49][V] InputReader: opening /dev/input/event1
[00:43:49][I] InputReader: btn:"qpnp_pon"
[00:43:49][V] InputReader: opening /dev/input/event3
[... 206 lines cut ...]
[00:50:32][V] AffinityHelper: UpdateTop(size=0/228) took 0.8/0.0/0.8ms
[00:50:32][V] AffinityHelper: UpdateFg(size=2/1738) took 3.6/1.0/4.6ms
[00:50:34][V] AffinityHelper: UpdateTop(size=0/15) took 4.4/0.0/4.4ms
[00:50:34][V] AffinityHelper: UpdateTop(size=0/15) took 0.2/0.0/0.2ms
[00:50:34][V] AffinityHelper: UpdateTop(size=0/14) took 0.2/0.0/0.3ms
[00:50:34][V] AffinityHelper: UpdateTop(size=0/13) took 0.2/0.0/0.2ms
[00:50:34][V] AffinityHelper: UpdateFg(size=3/127) took 0.4/0.7/1.1ms
[00:50:34][V] TopappMonitor: cannot get top-app name from dumpsys
[00:50:35][V] TopappMonitor: cannot get top-app name from dumpsys
[00:50:39][V] AffinityHelper: UpdateTop(size=0/13) took 0.3/0.0/0.3ms
[00:50:39][V] AffinityHelper: UpdateTop(size=0/13) took 0.4/0.0/0.4ms
[00:50:39][V] AffinityHelper: UpdateTop(size=0/13) took 0.2/0.0/0.2ms
[00:50:39][V] AffinityHelper: UpdateFg(size=23/96) took 0.2/4.2/4.4ms
[00:50:40][V] TopappMonitor: cannot get top-app name from dumpsys
[00:50:41][V] TopappMonitor: cannot get top-app name from dumpsys
[00:50:41][V] InputReader: pressed=1, predict=-1ms
[00:50:41][V] State: None(10257ms) -> Touch
[00:50:42][V] InputReader: pressed=0, predict=-1ms
[00:50:42][V] State: Touch(425ms) -> None

```
</details>

3. 沿用7.25版本，覆盖`libuperf.sh `和`powercfg_once.sh`内容后刷入能正常使用。
以上内容供yc参考，感谢！

TanZhiwen2001, 2021-10-18:
k20p也会

railjty, 2022-04-04:
将在uperf v3修复

closed 2023-01-19 by sino1641