yc9559/uperf #91: SsAnalysis导致蓝牙耳机无声音

suroot66, 2022-04-08:
如题，关闭或卸载SsAnalysis后重启**两次**就能恢复正常，而一旦启用，就会导致蓝牙耳机无媒体声音（通话声音或者外放还是有的，唯独听音乐玩游戏等媒体播放时没有声音），音量条可以调节但是无声。之前在cutoolbox（ct调度）中遇到过同样的问题，当初是安装了屏蔽用户态boosts模块导致的，也是需要删除模块后重启两次才能恢复，不知两者之间是否事出同因。

suroot66, 2022-04-08:
系统：安卓10 魔趣 骁龙855
magisk 24
yc调度正常启用

yc9559, 2022-04-09:
DEV 22.04.09版本暂时不用SsAnalysis，试试看

railjty, 2022-04-09:
> DEV 22.04.09版本暂时不用SsAnalysis，试试看

Github未发布？

suroot66, 2022-04-09:
好耶！谢谢yc大佬！😘💕

railjty, 2022-04-09:
> 好耶！谢谢yc大佬！😘💕

建议关闭问题

suroot66, 2022-04-09:
大佬！这回彻底gg了（捂脸
uperf也用不了了，一用蓝牙就没声了hhhh
可能是我手机比较特别？  需不需要日志啥的，我可以提供
总之现在暂时就没法用这个调度了hh😂

railjty, 2022-04-09:
> Reference in new i

安卓10 建议回退到v2

suroot66, 2022-04-09:
嗯，也是个办法

yc9559, 2022-04-09:
@suroot66 看看`/data/adb/modules/uperf/logcat.log`

FLY-xiu, 2022-04-11:
[logcat.log](https://github.com/yc9559/uperf/files/8461191/logcat.log)
同样的问题

railjty, 2022-04-11:
> [logcat.log](https://github.com/yc9559/uperf/files/8461191/logcat.log) 同样的问题

[image]
[image]
[image]
[image]
[image]
建议自行检查系统问题

suroot66, 2022-04-11:
因为之前魔趣变砖了，所以从魔趣回到了官方flyme系统
目前【22.04.09】在flyme9上已经没有该问题，一切运行正常。
推测原问题可能是第三方类原生ROM精简掉了一些东西导致的 （0 _0）