yc9559/uperf #96: 求助QAQ 卸载22.04.09之后无法启动任何app

suroot66, 2022-04-09:
刚magisk里卸载了uperf（卸载前是开着的），然后重新启动，结果什么app都打不开了，连桌面启动器app都一直加载中，通过状态栏快捷卡片打开任何app也没反应，不过我通过状态栏能够进入设置，开发者选项里面查看当前运行的进程只有设置、Android Services Library，缓存中的进程有电话、运营商默认应用、android core apps，其他的就什么都没有了，桌面一直进不去，通过adb命令安装了个第三方桌面然后设为默认桌面，依旧进不去。音量键似乎没反应，按电源键能锁屏，但锁屏后再按电源键却不能唤醒..
然后每隔一小会儿，系统UI就会重启
-------
再然后，我把sfa、ssa(上个版本的残留)都卸载了（通过mm救砖模块），依然不行，然后我干脆卸载了magisk，还是不行..我又平刷了一遍我现在用的ROM，还是没用，现在我已经束手无策了呜呜  大佬们知道这大概是什么原因吗，会是哪些配置残留的影响吗？
我不想双清呜呜我有几百个app还有桌面布局没有备份QAQ 
求求各位大佬给点建议，谢谢大家了！

suroot66, 2022-04-10:
确定是data分区出问题了，清空system分区和cache分区重刷rom也不行，又清掉了data分区，重刷，这次恢复了
我不太懂，目前猜测可能是模块卸载的过程出了问题，magisk应该是重启自动执行卸载代码吧，但是我的手机重启按钮有个bug，系统里点到”重启到recovery模式“时会无限重启，手动长按电源键才能恢复。当初我卸载uperf时手贱不小心点到了这个按钮，可能是这个非正常启动过程中导致卸载uperf出现了问题，于是mm模块里面名义上显示已卸载，实际上可能有哪里寄了。所以光magisk也没法恢复了，只能清掉data分区

------
玩机一时爽，救砖火葬场哈哈哈    有些按钮真的不能瞎按
不过刚好一直懒得刷机来着，一不做二不休，我打算折腾一下最新版的ROM了（

railjty, 2022-04-11:
建议只修改boot.img

yc9559, 2022-04-11:
uperf从v2后期版本开始，有开机60s内强制重启后自动关闭模块，以后再遇到卡开机可尝试这个

closed 2022-04-11 by yc9559