yc9559/uperf #99: 红米K30pro无法正常使用uperf

hosea666, 2022-04-15:
使用后大概率卡米
（有时候在我试探性的关闭一些其他模块后，可以正常重启，但是什么都不干第二次重启也可能会卡米）
系统是开发板公测最新版安卓12

hosea666, 2022-04-15:
23:35:37 I Uperf v3(22.04.09)[7667aa2], by Matt Yang (yccy@outlook.com)
23:35:37 I Config 'sdm865/sdm865+/sdm870[22.04.09]' by 'yc@coolapk'
23:35:37 I CpuGovernor cluster0:
23:35:37 I opp  614400 pwr 0.047 cost 0.067
23:35:37 I opp 1075200 pwr 0.089 cost 0.072
23:35:37 I opp 1171200 pwr 0.105 cost 0.078
23:35:37 I opp 1420800 pwr 0.157 cost 0.096
23:35:37 I opp 1516800 pwr 0.191 cost 0.110
23:35:37 I opp 1612800 pwr 0.230 cost 0.124
23:35:37 I opp 1708800 pwr 0.274 cost 0.139
23:35:37 I opp 1804800 pwr 0.323 cost 0.155
23:35:37 I CpuGovernor cluster1:
23:35:37 I opp  710400 pwr 0.121 cost 0.057
23:35:37 I opp 1056000 pwr 0.191 cost 0.060
23:35:37 I opp 1382400 pwr 0.327 cost 0.079
23:35:37 I opp 1766400 pwr 0.554 cost 0.105
23:35:37 I opp 1862400 pwr 0.650 cost 0.116
23:35:37 I opp 2054400 pwr 0.872 cost 0.141
23:35:37 I opp 2246400 pwr 1.140 cost 0.169
23:35:37 I opp 2419200 pwr 1.424 cost 0.196
23:35:37 I CpuGovernor cluster2:
23:35:37 I opp  960000 pwr 0.248 cost 0.086
23:35:37 I opp 1401600 pwr 0.362 cost 0.086
23:35:37 I opp 1747200 pwr 0.563 cost 0.107
23:35:37 I opp 2265600 pwr 0.975 cost 0.143
23:35:37 I opp 2361600 pwr 1.104 cost 0.156
23:35:37 I opp 2553600 pwr 1.396 cost 0.182
23:35:37 I opp 2649600 pwr 1.559 cost 0.196
23:35:37 I opp 2841600 pwr 1.923 cost 0.226
23:35:37 I Loading perapp rule
23:35:37 I Preset 'balance' -> default
23:35:37 I Preset 'balance' -> offscreen
23:35:37 I Preset 'fast' -> 'com.miHoYo.Yuanshen'
23:35:37 I Preset inode -> 'balance'
23:35:37 I Uperf is running
23:35:37 I SfAnalysis connected
23:35:46 I Preset inode -> 'fast'
23:36:55 I Preset inode -> 'balance'
23:37:12 I Terminated by user
23:37:12 I Terminated by user

这是log

hosea666, 2022-04-15:
好像我重启很多次uperf就会自己关掉然后就可以正常开机了

hosea666, 2022-04-16:
重启了上百次，还是卡米

hosea666, 2022-04-16:
按照酷安网友的说法，加了一行 sleep 5 ，疑似问题已解决

yc9559, 2022-04-17:
谢谢反馈，下一版会换个思路彻底解决这个问题😋

closed 2022-04-17 by yc9559

sino1641, 2023-02-26:
> 谢谢反馈，下一版会换个思路彻底解决这个问题😋

k30s 去年最新的miui开发版+0904的uperf 也是这个问题 log也是同样的 请问有什么办法可以解决吗