yc9559/uperf #117: 待機無法降至低頻，以及調整limit power無作用

xup61069, 2022-05-03:
您好，8g1小米12最低頻率小中大核分別只會到614,768,940MHz，和780g小米11lite 691,940,1094MHz，
如果能調用更低頻率能否更省電？
以及編輯uperf.json的slow和fast limit power，看起來沒什麼作用，是否和boost機制有關，調整哪個參數可以更好的強制限制功耗呢？
謝謝！

xup61069, 2022-05-03:
爬酷安找到改freefreq就是改最低頻率，改完發熱量明顯降低，應該是這樣做讓power limit正常運作了，對流暢度也不影響，不知道原始最低頻率較高的用意是什麼
[image]
[image]

wuc656, 2022-05-04:
> 爬酷安找到改freefreq就是改最低頻率，順帶改了plain和sweet，改完發熱量明顯降低，應該是這樣做讓power limit正常運作了，對流暢度似乎也不影響，不知道原始最低頻率較高的用意是什麼，希望到時候有v3的詳細說明，謝謝！ [image]

可以分享一下你的 uperf.json 嗎?

xup61069, 2022-05-04:
> 可以分享一下你的 uperf.json 嗎?
https://jsoneditoronline.org/#left=cloud.9b9a818005ac4d9c850b8ba8547032fc
只有改上面提到的數值
[image]
[image]

yc9559, 2022-05-07:
改最低频率不一定有正向收益，EAS开发有人提过这个想法。发热降低主要是slowlimit带来的。😋