yc9559/uperf #120: 支持schedhorizon调度器

ajdgsu, 2022-05-04:
我的手机默认的cpu调度器是schedhorizon，实际上就是schedutil的魔改版本，在schedutil的基础上增加了两个参数。最新版的uperf显示不支持此平台 (之前没有问题) 因此完全可以把schedutil的参数放进去而不是直接停止。

日志在此
19:25:55 I Uperf v3(22.05.04)[172d1b06], by Matt Yang (yccy@outlook.com)
19:25:55 I Config 'sdm865/sdm865+/sdm870[22.05.04]' by 'yc@coolapk'
19:25:55 I Current home is 'com.sec.android.app.launcher'
19:25:55 E Tried CpufreqWriterEpic: Failed to open epic minfreq knob for cluster0
19:25:55 E Tried CpufreqWriterPpm: Failed to open MTK PPM hard userlimit minfreq knob
19:25:55 E Cpufreq target 1075200 actual 1075200
19:25:55 E Cpufreq target 1670400 actual 1286400
19:25:55 E Tried CpufreqWriterCap: Failed to write freq
19:25:55 E Exception thrown: No CpufreqWriter supported for this platform
19:25:55 I Failed to start uperf(pid=12839)

ajdgsu, 2022-05-05:
抱歉，之前没看到，这才是日志
[logcat.zip](https://github.com/yc9559/uperf/files/8628453/logcat.zip)

yc9559, 2022-05-07:
自检没过一般不是魔改的调速器不支持。下一版会回退混合调频😋

closed 2022-05-07 by yc9559