yc9559/uperf #121: Failed to activate Uperf. MTK Helio G90T (MT6785)

EdelStark, 2022-05-05:
Last versions was perfectly fine. Please support Helio G90T MT6785. I think it only supports G90T with MT6768

19:44:16 I Uperf v3(22.05.04)[172d1b06], by Matt Yang (yccy@outlook.com)
19:44:16 I Config 'Helio G90T(mt6768)[22.05.04]' by 'yc@coolapk'
19:44:16 I Current home is 'com.google.android.apps.nexuslauncher'
19:44:16 E Tried CpufreqWriterEpic: Failed to open epic minfreq knob for cluster0
19:44:16 E Cpufreq target 1500000 actual 2000000
19:44:16 E Tried CpufreqWriterPpm: Failed to write freq
19:44:16 E Cpufreq target 1500000 actual 2000000
19:44:16 E Tried CpufreqWriterCap: Failed to write freq
19:44:16 E Exception thrown: No CpufreqWriter supported for this platform
19:44:16 I Failed to start uperf(pid=7010)

railjty, 2022-05-07:
请尝试重启机器？1200设备无法复现此问题

EdelStark, 2022-05-07:
> 

Already tried. Sometimes works but sometimes not. It works perfectly when it runs but after a reboot, it suddenly won't activate.

railjty, 2022-05-07:
> > 
> 
> Already tried. Sometimes works but sometimes not. It works perfectly when it runs but after a reboot, it suddenly won't activate.

Try to use 22.04.30-Fix1？

EdelStark, 2022-05-07:
> > > 
> > 
> > 
> > Already tried. Sometimes works but sometimes not. It works perfectly when it runs but after a reboot, it suddenly won't activate.
> 
> Try to use 22.04.30-Fix1？

Yep, used it. But 22.05.04 update was the most stable so far in my chipset. The only thing is the deactivation of module everytime i reboot. Any possible fixes? Not complaining tho. I just love uperf so much.

railjty, 2022-05-07:
> Yep, used it. But 22.05.04 update was the most stable so far in my chipset. The only thing is the deactivation of module everytime i reboot. Any possible fixes? Not complaining tho. I just love [...]

Try install 22.05.04 and when it reports error, do `ls -la /proc/ppm/policy` and upload the result

EdelStark, 2022-05-07:
> 

Will do. I'll send the result here. Let me reflash the module again and i'll try to reoccur the issue.

EdelStark, 2022-05-07:
~ $ su
s -la /proc/ppm/policy                                <
total 0
dr-xr-xr-x 2 root   root   0 2022-05-07 21:56 .
dr-xr-xr-x 5 root   root   0 2022-05-07 21:56 ..
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 forcelimit_cpu_core
-rw-rw-r-- 1 system system 0 2022-05-07 21:56 hard_userlimit_cpu_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 hard_userlimit_freq_limit_by_others
-r--r--r-- 1 root   root   0 2022-05-07 21:56 hard_userlimit_max_cpu_freq
-r--r--r-- 1 root   root   0 2022-05-07 21:56 hard_userlimit_min_cpu_freq
-rw-rw-r-- 1 root   root   0 2022-05-07 21:56 lcmoff_min_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 ptpod_test
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 sysboost_cluster_freq_limit
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 sysboost_cluster_freqidx_limit
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 sysboost_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 thermal_cur_power
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 thermal_limit
-rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_cpu_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 userlimit_freq_limit_by_others
-rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_max_cpu_freq
-rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_min_cpu_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 ut_fix_core_num
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 ut_fix_freq_idx
:/data/data/com.termux/files/home #119

EdelStark, 2022-05-07:
It happened again. Here's the log.

~ $ su
s -la /proc/ppm/policy                                <
total 0
dr-xr-xr-x 2 root   root   0 2022-05-07 21:56 .
dr-xr-xr-x 5 root   root   0 2022-05-07 21:56 ..
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 forcelimit_cpu_core
-rw-rw-r-- 1 system system 0 2022-05-07 21:56 hard_userlimit_cpu_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 hard_userlimit_freq_limit_by_others
-r--r--r-- 1 root   root   0 2022-05-07 21:56 hard_userlimit_max_cpu_freq
-r--r--r-- 1 root   root   0 2022-05-07 21:56 hard_userlimit_min_cpu_freq
-rw-rw-r-- 1 root   root   0 2022-05-07 21:56 lcmoff_min_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 ptpod_test
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 sysboost_cluster_freq_limit
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 sysboost_cluster_freqidx_limit
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 sysboost_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 thermal_cur_power
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 thermal_limit
-rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_cpu_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 userlimit_freq_limit_by_others
-rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_max_cpu_freq
-rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_min_cpu_freq
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 ut_fix_core_num
-rw-r--r-- 1 root   root   0 2022-05-07 21:56 ut_fix_freq_idx
:/data/data/com.termux/files/home #

railjty, 2022-05-07:
> ~ $ su s -la /proc/ppm/policy < total 0 dr-xr-xr-x 2 root root 0 2022-05-07 21:56 . dr-xr-xr-x 5 root root 0 2022-05-07 21:56 .. -rw-r--r-- 1 root root 0 2022-05-07 21:56 forcelimit_cpu_core [...]

It seems some program or system locked  hard_userlimit_max_cpu_freq and  hard_userlimit_min_cpu_freq. Please unlock it first

EdelStark, 2022-05-07:
> It seems some program or system locked hard_userlimit_max_cpu_freq and hard_userlimit_min_cpu_freq. Please unlock it first

Any chances on how to unlock it? I didn't do anything on my system. I'm just using uperf eversince and all of the update never had issues on mine except on the latest 22.05.04 update.

railjty, 2022-05-07:
> Any chances on how to unlock it? I didn't do anything on my system. I'm just using uperf eversince and all of the update never had issues on mine except on the latest 22.05.04 update
chmod 666 /proc/ppm/policy/* ，and run service.sh in /data/adb/modules/uperf

EdelStark, 2022-05-07:
> `chmod 666 /proc/ppm/policy/* `，and run service.sh in /data/adb/modules/uperf

I guess this commands fixed my issue. Let me observe a little bit more if it doesn't activate every boot.

railjty, 2022-05-07:
https://github.com/yc9559/uperf/blob/75c8079d2d0fe6789d8b4562848446bb640a3218/magisk/script/powercfg_once.sh#L146
Delete this row and row 147, and try again

EdelStark, 2022-05-07:
> https://github.com/yc9559/uperf/blob/75c8079d2d0fe6789d8b4562848446bb640a3218/magisk/script/powercfg_once.sh#L146-L147
> 
> 
> Delete these rows, and try again
> > > > > > ~ $ su s -la /proc/ppm/policy < total 0 dr-xr-xr-x 2 root root 0 2022-05-07 21:56 . dr-xr-xr-x 5 root root 0 2022-05-07 21:56 .. -rw-r--r-- 1 root root 0 2022-05-07 21:56 forcelimit_cpu_core -rw-rw-r-- 1 system system 0 2022-05-07 21:56 hard_userlimit_cpu_freq -rw-r--r-- 1 root root 0 2022-05-07 21:56 hard_userlimit_freq_limit_by_others -r--r--r-- 1 root root 0 2022-05-07 21:56 hard_userlimit_max_cpu_freq -r--r--r-- 1 root root 0 2022-05-07 21:56 hard_userlimit_min_cpu_freq -rw-rw-r-- 1 root root 0 2022-05-07 21:56 lcmoff_min_freq -rw-r--r-- 1 root root 0 2022-05-07 21:56 ptpod_test -rw-r--r-- 1 root root 0 2022-05-07 21:56 sysboost_cluster_freq_limit -rw-r--r-- 1 root root 0 2022-05-07 21:56 sysboost_cluster_freqidx_limit -rw-r--r-- 1 root root 0 2022-05-07 21:56 sysboost_freq -rw-r--r-- 1 root root 0 2022-05-07 21:56 thermal_cur_power -rw-r--r-- 1 root root 0 2022-05-07 21:56 thermal_limit -rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_cpu_freq -rw-r--r-- 1 root root 0 2022-05-07 21:56 userlimit_freq_limit_by_others -rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_max_cpu_freq -rw-rw---- 1 system system 0 2022-05-07 21:56 userlimit_min_cpu_freq -rw-r--r-- 1 root root 0 2022-05-07 21:56 ut_fix_core_num -rw-r--r-- 1 root root 0 2022-05-07 21:56 ut_fix_freq_idx :/data/data/com.termux/files/home #119
> > > > > 
> > > > > 
> > > > > It seems some program or system locked hard_userlimit_max_cpu_freq and hard_userlimit_min_cpu_freq. Please unlock it first
> > > > 
> > > > 
> > > > Any chances on how to unlock it? I didn't do anything on my system. I'm just using uperf eversince and all of the update never had issues on mine except on the latest 22.05.04 update
> > > 
> > > 
> > > `chmod 666 /proc/ppm/policy/* `，and run service.sh in /data/adb/modules/uperf
> > 
> > 
> > I guess this commands fixed my issue. Let me observe a little bit more if it doesn't activate every boot.

Already deleted the lines. Let me check if it still occurs after a few reboots. Will give feedback to you later.

EdelStark, 2022-05-07:
Sadly. It still occurs. Any other troubleshoots or fixes? Also a question, does clean flashing the system will fix this issue?

railjty, 2022-05-07:
> Sadly. It still occurs. Any other troubleshoots or fixes? Also a question, does clean flashing the system will fix this issue?

`chmod 666 /proc/ppm/policy/* ` Add this to https://github.com/yc9559/uperf/blob/75c8079d2d0fe6789d8b4562848446bb640a3218/magisk/script/powercfg_once.sh#L146-L147

EdelStark, 2022-05-07:
> 

Already deleted these lines. I also ran the command. Let me check again and will give feedback to you.

EdelStark, 2022-05-07:
It's still the same after a few reboots.

EdelStark, 2022-05-07:
Update: retried the fixes you've mentioned. And after a few reboot, the config does stick and is working. Let me check a little bit more.

EdelStark, 2022-05-07:
I have one last question. Does reflashing the whole system can fix the locked state of hard_userlimit_max_cpu_freq and hard_userlimit_min_cpu_freq? So i can clean my system to make uperf work without any troubleshootings.

railjty, 2022-05-08:
> I have one last question. Does reflashing the whole system can fix the locked state of hard_userlimit_max_cpu_freq and hard_userlimit_min_cpu_freq? So i can clean my system to make uperf work without any troubleshootings.

Maybe delete all modules and other userspace schedulers (like CuToolbox, Femind, Scene 5's builtin scheduler) will be better?

EdelStark, 2022-05-08:
> Maybe delete all modules and other userspace schedulers (like CuToolbox, Femind, Scene 5's builtin scheduler) will be better?

Let me reinstall scene. I guess that app was the culprit. But when i turn on the turned off PPM, it reverts back after a reboot. I don't know what's happening.

EdelStark, 2022-05-08:
> 

I guess this is the culprit. But whenever i turn on again the parameters, it reverted back to off. 

https://drive.google.com/file/d/1cQRWMeu8W2jUlBHKWXZlZpAwEm3uYzyF/view?usp=drivesdk

railjty, 2022-05-08:
> > 
> 
> I guess this is the culprit. But whenever i turn on again the parameters, it reverted back to off.
> 
> https://drive.google.com/file/d/1cQRWMeu8W2jUlBHKWXZlZpAwEm3uYzyF/view?usp=drivesdk

Because these settings are not permanent. I think the Scene works correctly

EdelStark, 2022-05-08:
> > > 
> > 
> > 
> > I guess this is the culprit. But whenever i turn on again the parameters, it reverted back to off.
> > https://drive.google.com/file/d/1cQRWMeu8W2jUlBHKWXZlZpAwEm3uYzyF/view?usp=drivesdk
> 
> Because these settings are not permanent. I think the Scene works correctly

I see. Is there anything that i can do to unlock this properly? The later fixes never worked for me. Weirdest part is, everytime i went into reboot, sometimes it works, sometimes not. It looks like those PPMs that you've mentioned were interferring on uperf and i don't know how did i locked it.

railjty, 2022-05-08:
[powercfg_once.txt](https://github.com/yc9559/uperf/files/8646685/powercfg_once.txt)
用这个替换powercfg_once.sh看看？

EdelStark, 2022-05-08:
> [powercfg_once.txt](https://github.com/yc9559/uperf/files/8646685/powercfg_once.txt) 用这个替换powercfg_once.sh看看？

Still doesn't work. My friend gave a feedback to me. The latest update also doesn't work with him. But all versions of uperf doesn't have this problem. Only this update is. He also has MTK G90T (MT6785) chipset.

railjty, 2022-05-08:
> Still doesn't work. My friend gave a feedback to me. The latest update also doesn't work with him. But all versions of uperf doesn't have this problem. Only this update is. He also has MTK G90T [...]

Maybe it's a strange problem caused by uperf.

EdelStark, 2022-05-08:
> Maybe it's a strange problem caused by uperf.

Many of us use uperf since last year. It don't have any issues like this since day 1. Maybe there's something that needs to be fixed. Btw, i'll wait for the next update. Hope that this can be fixed by the next update. Thank you.

railjty, 2022-05-08:
> Many of us use uperf since last year. It don't have any issues like this since day 1. Maybe there's something that needs to be fixed. Btw, i'll wait for the next update. Hope that this can be fixed [...]

Sorry, but I'm not the maintainer. The maintainer is only one, @yc9559 herself.

railjty, 2022-05-09:
> Many of us use uperf since last year. It don't have any issues like this since day 1. Maybe there's something that needs to be fixed. Btw, i'll wait for the next update. Hope that this can be fixed [...]

Try 22.05.09?

EdelStark, 2022-05-09:
> Try 22.05.09?

Yep. And it after 2 consecutive reboots, it is working well. However, i noticed that my CPU governor for big clusters went from schedutil to powersave. Why is that?

railjty, 2022-05-09:
> Yep. And it after 2 consecutive reboots, it is working well. However, i noticed that my CPU governor for big clusters went from schedutil to powersave. Why is that?

See changelog

EdelStark, 2022-05-09:
> See changelog

Yep. I saw the changelog. I'm just confused. Last update were both schedutil but uperf doesn't work properly. Maybe this is proper? Does the cpu governor for big cluster go to schedutil when i play game? Or it sticks to powersave permanently?

railjty, 2022-05-09:
> Yep. I saw the changelog. I'm just confused. Last update were both schedutil but uperf doesn't work properly. Maybe this is proper? Does the cpu governor for big cluster go to schedutil when i play [...]

Uperf V3 used powersave and change the frequency by itself to avoid the changes made by system

EdelStark, 2022-05-09:
> Uperf V3 used powersave and change the frequency by itself to avoid the changes made by system

Oh. Now i understand. Using the latest update right now and i must say that the issue were fixed. But i'll take my time to check and monitor if it still fails. For now, it is working perfectly. Also a question, does custom kernel interferes with uperf presets? Does it better to use stock kernel of the system while using uperf?

railjty, 2022-05-09:
> 

The author only use stock kernel currently. Other OSS kernels are not tested

EdelStark, 2022-05-22:
Hello, it is me again. Uperf is working fine as of latest update on my processor. I have a question, how can i use this? And how do i implement this?

Preset Performance Mode】

Caton powersave: Large performance limit, suitable for users who do not require high fluency
Balanced balance: Moderate performance limit, suitable for daily mobile phone
Power consumption performance: approximately equal to no performance limit, suitable for tablet daily life
Extremely fast: similar to the cost of electricity, with continuous performance output, suitable for mobile gaming

Thank you.

EdelStark, 2022-05-24:
> Write the desired powermode to /storage/emulated/0/Android/yc/uperf/cur_powermode.txt 😋

Do i need to type in chinese language? Because when i try to put it in english language, uperf doesn't recognize the powermode.

yc9559, 2022-05-24:
Type powersave😋

railjty, 2022-05-24:
Execute `sh /data/powercfg.sh powersave`