yc9559/uperf #130: DEV 22.05.20 Failed to start uperf

wukibaka, 2022-05-20:
一加5T
PE12（安卓12.1）
[logs.zip](https://github.com/yc9559/uperf/files/8739069/logs.zip)

railjty, 2022-05-20:
[image]
三方内核？还是说这个PE自带调度？

wukibaka, 2022-05-20:
> [image]
> 三方内核？还是说这个PE自带调度？

第三方内核

degr15, 2022-05-22:
[image]

I have the same problem with sd732G. I'm using crdroid. This problem doesn't happen on DEV 22.05.09

yc9559, 2022-05-23:
Will be fixed in the next dev version.😋

closed 2022-07-08 by yc9559