yc9559/uperf #133: Abnormal idle power usage (Mi 9T Pro, SD855, uperf 20.05.20)

Kugumin, 2022-05-27:
So as per title... Without uperf device on idle can reach as low as 80mA power usage even with boosting enabled and even with higher frequencies than uperf sets. After installing uperf lowest power usage I get is 200mA. On older version it was possible to get lower, like 120-150 mA.

I used everything latest:
Uperf
SSA
SFA

And still power usage is higher than it should be.

Maybe some power config adjustments should be made to improve the situation for SD855.

Anyways, uperf is very interesting module to improve fluency, but in my opinion idle usage should be adjusted for even better user experience!

Thanks for attention.

yc9559, 2022-05-30:
Thank you for your feedback. sdm845-865 standby issue will be fixed in dev 22.06.01😋

Kugumin, 2022-06-05:
Tried new version. Idle is good now! There's a room to improve in such apps like TikTok, but it's already good enough. Thank you very much!

closed 2022-06-09 by yc9559