yc9559/uperf #152: [Bug Report] Failed Write Minfreq

URARUA, 2022-07-25:
Device: Oneplus 7T Pro running OOS 11
Only working on official kernel, tried [Dora](https://github.com/nem0-z/dora_kernel_oneplus_sm8150/tree/eleven) and [Blu-spark](https://github.com/engstk/op7) , none of them working.
log here:
`13:56:31 I uperf v3(22.07.24)[c4ae5755], by Matt Yang (yccy@outlook.com)
13:56:31 I Config 'sdm855/sdm855+/sdm860[22.07.24]' by 'yc@coolapk'
13:56:31 I Current home is 'com.teslacoilsw.launcher'
13:56:32 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
13:56:32 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
13:56:32 E Tried CpufreqWriterPpmFixed: Failed to open MTK PPM hard userlimit minfreq knob
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterMsmPowersave: Failed to write freq
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterPowersave: Failed to write freq
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterMsmFixed: Failed to write freq
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterPerformance: Failed to write freq
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterFixed: Failed to write freq
13:56:32 E Retrying...
13:56:32 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
13:56:32 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
13:56:32 E Tried CpufreqWriterPpmFixed: Failed to open MTK PPM hard userlimit minfreq knob
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterMsmPowersave: Failed to write freq
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterPowersave: Failed to write freq
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterMsmFixed: Failed to write freq
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterPerformance: Failed to write freq
13:56:32 E Cpufreq target 1113600 actual 1785600
13:56:32 E Tried CpufreqWriterFixed: Failed to write freq
13:56:32 E Exception thrown: No CpufreqWriter supported for this platform
`

SiMplTon-B, 2022-07-30:
Me Too ! 
Device:SAMSUNG Tab S6 T860

Fouread, 2022-08-11:
redmi k20p, Android 12, 类原生系统，同样问题

日志：
07:34:51 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
07:34:51 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
07:34:51 E Tried CpufreqWriterPpmFixed: Failed to open MTK PPM hard userlimit minfreq knob
07:34:51 E Tried CpufreqWriterMsmPowersave: Failed to open msm minfreq knob
07:34:51 E Tried CpufreqWriterPowersave: Failed to set powersave governor
07:34:51 E Tried CpufreqWriterMsmFixed: Failed to open msm minfreq knob
07:34:51 E Cpufreq target 1305600 actual 1708800
07:34:51 E Tried CpufreqWriterPerformance: Failed to write freq
07:34:51 E Cpufreq target 1305600 actual 1708800
07:34:51 E Tried CpufreqWriterFixed: Failed to write freq
07:34:51 E Exception thrown: No CpufreqWriter supported for this platform

Puqns67, 2022-08-22:
realme gt neo flash edition, Android 12, RealmeUI 3.0(RMX3350_11_C.17), 相同故障

```
17:49:49 I uperf v3(22.07.24)[c4ae5755], by Matt Yang (yccy@outlook.com)
17:49:49 I Config 'Dimensity 1200(mt6893)/1100(mt6891z)[22.07.24]' by 'yc@coolapk'
17:49:49 I Current home is 'com.android.launcher'
17:49:50 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
17:49:50 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
17:49:50 E Cpufreq target 1250000 actual 2000000
17:49:50 E Tried CpufreqWriterPpmFixed: Failed to write freq
17:49:50 E Tried CpufreqWriterMsmPowersave: Failed to open msm minfreq knob
17:49:50 E Cpufreq target 1250000 actual 2000000
17:49:50 E Tried CpufreqWriterPowersave: Failed to write freq
17:49:50 E Tried CpufreqWriterMsmFixed: Failed to open msm minfreq knob
17:49:50 E Cpufreq target 1250000 actual 2000000
17:49:50 E Tried CpufreqWriterPerformance: Failed to write freq
17:49:50 E Cpufreq target 1250000 actual 2000000
17:49:50 E Tried CpufreqWriterFixed: Failed to write freq
17:49:50 E Retrying...
17:49:50 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
17:49:50 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
17:49:50 E Cpufreq target 1250000 actual 2000000
17:49:50 E Tried CpufreqWriterPpmFixed: Failed to write freq
17:49:50 E Tried CpufreqWriterMsmPowersave: Failed to open msm minfreq knob
17:49:50 E Cpufreq target 1250000 actual 2000000
17:49:50 E Tried CpufreqWriterPowersave: Failed to write freq
17:49:50 E Tried CpufreqWriterMsmFixed: Failed to open msm minfreq knob
17:49:50 E Cpufreq target 1250000 actual 2000000
17:49:50 E Tried CpufreqWriterPerformance: Failed to write freq
17:49:50 E Cpufreq target 1250000 actual 2000000
17:49:50 E Tried CpufreqWriterFixed: Failed to write freq
17:49:50 E Exception thrown: No CpufreqWriter supported for this platform
17:49:50 I Failed to start uperf(pid=7425)
```

Younvkon, 2022-09-18:
魅族17类似故障

qwerzl, 2022-11-03:
Sony Xperia 1 IV, Android 12, Snapdragon 8gen1, 使用第三方内核时出现同样故障。
```
$ uname -a
Linux localhost 5.10.81-Kirisakura_Nagara_S_1.0.0-00037-gdca213822279-dirty #1 SMP PREEMPT Mon Aug 15 09:23:09 UTC 2022 aarch64
```
```
$ cat /sdcard/Android/yc/uperf/uperf_log.txt
16:23:30 I uperf v3(22.09.04)[acc04447], by Matt Yang (yccy@outlook.com)
16:23:30 I Config 'sdm8g1[22.09.04]' by 'yc@coolapk'
16:23:31 I Current home is 'shubh.ruthless'
16:23:31 I No available opp of cpu7, use generated opp instead
16:23:31 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
16:23:31 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
16:23:31 E Tried CpufreqWriterPpmFixed: Failed to open MTK PPM hard userlimit minfreq knob
16:23:31 E Tried CpufreqWriterMsmPowersave: Failed to set powersave governor
16:23:31 E Tried CpufreqWriterPowersave: Failed to open scaling min knob for cluster2
16:23:31 E Cpufreq target 1075200 actual 1075200
16:23:31 E Cpufreq target 1651200 actual 1651200
16:23:31 E Cpufreq target 1900000 actual 2515200
16:23:31 E Tried CpufreqWriterMsmFixed: Failed to write freq
16:23:31 E Tried CpufreqWriterPerformance: Failed to open scaling min knob for cluster2
16:23:31 E Tried CpufreqWriterFixed: Failed to open scaling min knob for cluster2
16:23:31 E Retrying...
16:23:31 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
16:23:31 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
16:23:31 E Tried CpufreqWriterPpmFixed: Failed to open MTK PPM hard userlimit minfreq knob
16:23:31 E Tried CpufreqWriterMsmPowersave: Failed to set powersave governor
16:23:31 E Tried CpufreqWriterPowersave: Failed to open scaling min knob for cluster2
16:23:31 E Cpufreq target 1075200 actual 1075200
16:23:31 E Cpufreq target 1651200 actual 1651200
16:23:31 E Cpufreq target 1900000 actual 2515200
16:23:31 E Tried CpufreqWriterMsmFixed: Failed to write freq
16:23:31 E Tried CpufreqWriterPerformance: Failed to open scaling min knob for cluster2
16:23:31 E Tried CpufreqWriterFixed: Failed to open scaling min knob for cluster2
16:23:31 E Exception thrown: No CpufreqWriter supported for this platform
16:23:31 I Failed to start uperf(pid=13342)
```