yc9559/uperf #172: Support for SD 720g Processor

AhmadMahdee, 2022-10-29:
sir can u add sd 720g soc support in your uperf module???
It would be great if u add it!!

DaveRamirezCodes, 2022-11-10:
Bump!
First of all thank you for your hard work ... I follow the evolution of the project since the times of project wipe from the days of project wipe, at the time I had a Snapdragon 652 device, And thanks to your project I was able to squeeze the good old 652 thoroughly ...

I would like to be able to continue enjoying your amazing project with my new redmi note 9s too

My device is ro.board.platform=ATOLL , ro.product.board=CURTANA, and when I install UPERF module , The installation fails saying "target [CURTANA] is not supported

So sir please ...can you add support for snapdragon 720g (SoC sm7125, Codename: "Atoll", For Xiaomi devices with this soc it is sometimes also called "Miatoll" from the merger of "mi" for "xiaomi" and "atoll" for board codename)?... it is very similar to the supported snapdragon 730, they also have the same GPU (adreno 618). If you need testers for the Atoll/Miatoll family of devices I'm available... Thanks in advance.

GalaxyNight, 2022-11-17:
curtana already has an excellent kernel and roms, no need to use an AIO-tweaker like uperf.