yc9559/uperf #174: 使用uperf后玩节奏游戏时会出现stuttering现象。

BaeKILB, 2022-11-20:
正在使用翻译器 不好意思。
在玩名为"Project sekai"的游戏时,在0.1秒内会发生细微的断开。
uperf也知道有识别画面触摸操作的部分,所以用自动模式运行了游戏,结果没有发生这种现象。
我想问一下能不能解决。
或者uperf可以在不重新启动机器的情况下退出功能吗？
因为只有关闭magisk的uperf,重新启动设备,才不会发生这种现象。
感谢您的阅读。

I'm sorry I used a translator.
When playing a game called Project Sekai, a one-minute break occurs in 0.1 seconds.
This phenomenon did not occur when Superf learned that he also recognized the screen touch operation and ran the game in automatic mode.
Could you solve this problem by any chance?
Or can I shut down the upperf function without rebooting the device?
This is because this phenomenon does not occur only when the upper f of the magic mask is turned off and the equipment is rebooted.
Thank you for reading it.

Yan233th, 2022-11-21:
The same question, my phone is MI MIX2S, Snapdragon 845, MIUI 12.5.1, Android 10

BaeKILB, 2022-12-05:
> The same question, my phone is MI MIX2S, Snapdragon 845, MIUI 12.5.1, Android 10

I modified the uperf preset after looking at the uperf manual.

https://github.com/yc9559/uperf/tree/master/config

I modified the uperf setting with the aim of changing the cpu clock rapidly.
As a result, I could feel that the frame breaking problem was definitely getting better.

This is a preset for project sekai that I am using.
I'm trying to create a new preset, but there's an error, so I'm using the rarely used "powersave" preset instead.

Note that the uperf preset location is [storage/emulated/0/android/yc/superf]
The preset file is [superf.json], and the current preset can be set in [cur_powermode.txt].
Also, the file that sets the preset for each app in auto mode is [perapp_powermode.txt].

"powersave": {
      "*": {
        "cpu.latencyTime": 0.1, 
        "cpu.slowLimitPower": 5.0, 
        "cpu.fastLimitPower": 5.0, 
        "cpu.fastLimitCapacity": 64.0, 
        "cpu.margin": 0.1
      }, 
      "idle": {
        "cpu.baseSampleTime": 0.02, 
        "cpu.baseSlackTime": 0.04, 
        "cpu.predictThd": 0.3, 
        "sched.scene": "idle"
      }, 
      "touch": {
        "sched.scene": "touch"
      }, 
      "trigger": {
        "sched.scene": "touch"
      }, 
      "gesture": {
        "sched.scene": "touch"
      }, 
      "junk": {
        "cpu.burst": 0.6, 
        "sched.scene": "touch"
      }, 
      "switch": {
        "cpu.burst": 0.2, 
        "sched.scene": "boost"
      }
    },

closed 2022-12-05 by BaeKILB

Yan233th, 2022-12-05:
> I modified the uperf preset after looking at the uperf manual.
>
> https://github.com/yc9559/uperf/tree/master/config
>
> I modified the uperf setting with the aim of changing the cpu clock rapidly. As a [...]

Okay, I will try it later, thanks

Yan233th, 2022-12-07:
I have done it, thanks