yc9559/uperf #214: 我的手机CPU是天机700，安卓11的时候是能用的，但是更新系统之后就只报错

LiJiu-gfgfv, 2024-09-26:
[image]
[image]
报错内容：
23:03:32 I uperf v3(22.09.04)[acc04447], by Matt Yang (yccy@outlook.com)
23:03:32 I Config 'Dimensity 700(mt6833)/720(mt6853)/810(mt6833v)[22.09.04]' by 'yc@coolapk'
23:03:32 I Current home is 'com.android.launcher'
23:03:33 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
23:03:33 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
23:03:33 E Cpufreq target 1280000 actual 2000000
23:03:33 E Tried CpufreqWriterPpmFixed: Failed to write freq
23:03:33 E Tried CpufreqWriterMsmPowersave: Failed to open msm minfreq knob
23:03:33 E Cpufreq target 1280000 actual 2000000
23:03:33 E Tried CpufreqWriterPowersave: Failed to write freq
23:03:33 E Tried CpufreqWriterMsmFixed: Failed to open msm minfreq knob
23:03:33 E Cpufreq target 1280000 actual 2000000
23:03:33 E Tried CpufreqWriterPerformance: Failed to write freq
23:03:33 E Cpufreq target 1280000 actual 2000000
23:03:33 E Tried CpufreqWriterFixed: Failed to write freq
23:03:33 E Retrying...
23:03:33 E Tried CpufreqWriterEpicPowersave: Failed to open epic minfreq knob for cluster0
23:03:33 E Tried CpufreqWriterEpicFixed: Failed to open epic minfreq knob for cluster0
23:03:33 E Cpufreq target 1280000 actual 2000000
23:03:33 E Tried CpufreqWriterPpmFixed: Failed to write freq
23:03:33 E Tried CpufreqWriterMsmPowersave: Failed to open msm minfreq knob
23:03:33 E Cpufreq target 1280000 actual 2000000
23:03:33 E Tried CpufreqWriterPowersave: Failed to write freq
23:03:33 E Tried CpufreqWriterMsmFixed: Failed to open msm minfreq knob
23:03:33 E Cpufreq target 1280000 actual 2000000
23:03:33 E Tried CpufreqWriterPerformance: Failed to write freq
23:03:33 E Cpufreq target 1280000 actual 2000000
23:03:33 E Tried CpufreqWriterFixed: Failed to write freq
23:03:33 E Exception thrown: No CpufreqWriter supported for this platform

NCLnclNCL, 2024-12-23:
Reboot until it is successfully

LiJiu-gfgfv, 2024-12-23:
OK

LiJiu-gfgfv, 2024-12-23:
> Reboot until it is successfully 

我从安装这个面具模块开始重启了三遍，还是报错怎么办
I restarted it three times since I installed this mask module. What should I do if I still report an error

NCLnclNCL, 2024-12-23:
> > Reboot until it is successfully
> 
> 我从安装这个面具模块开始重启了三遍，还是报错怎么办 使用了APatch I restarted it three times since I installed this mask module. What should I do if I still report an error APatch was used.

Try this https://github.com/yinwanxi/Uperf-Game-Turbo

LiJiu-gfgfv, 2024-12-23:
> > > Reboot until it is successfully
> > 
> > 我从安装这个面具模块开始重启了三遍，还是报错怎么办 使用了APatch I restarted it three times since I installed this mask module. What should I do if I still report an error APatch was used.
> 
> Try this https://github.com/yinwanxi/Uperf-Game-Turbo

OK, I'll try我试试

LiJiu-gfgfv, 2024-12-23:
> > > Reboot until it is successfully
> > 
> > 我从安装这个面具模块开始重启了三遍，还是报错怎么办 使用了APatch I restarted it three times since I installed this mask module. What should I do if I still report an error APatch was used.
> 
> Try this https://github.com/yinwanxi/Uperf-Game-Turbo

依旧[image]