yc9559/uperf #216: Ksun安装sfanalysis后卡第一屏

XinSongalergy, 2025-02-05:
卡第一屏后用adb抓的log
[logcat.log](https://github.com/user-attachments/files/18677592/logcat.log)
希望作者大大修复一下，万分感谢

XinSongalergy, 2025-02-05:
顺带一提，该问题在apatch和ksu上也会出现

omajili-manbu, 2025-08-31:
嗨！虽然我不是模块作者，因为此模块导致开机卡在第一画面，很大可能是其post-fs-data脚本出现了问题，经过我的排查，代码有以下风险导致卡屏
1.
__set_perm $MODDIR/$SF 0 0 0755 "$(ls -Zl /$SF | cut -d' ' -f5)"
有可能给予surfaceflinger错误权限导致SELinux拒绝加载，权限设置不安全
2.
在boot早期触发的post-fs-data中使用patchelf风险很高，有可能patchelf依赖的动态库尚未加载完
试图对/system下的文件进行修改，极大可能被SELinux拒绝操作
3.
代码逻辑出错：
if [ -f "$MODDIR/flag/need_recuser" ]; then
    rm -f $MODDIR/flag/need_recuser
    true >$MODDIR/disable
    exit 0
else
    true >$MODDIR/flag/need_recuser
fi
第一次开机会执行本身就不稳定的patchelf，然后写入disable把自己关闭，此后开机时永不会再运行，这个问题在magisk30.x极其严重

修复了bug的sh脚本：
[post-fs-data.sh](https://github.com/user-attachments/files/22068826/post-fs-data.sh)

祝一切顺利！

XinSongalergy, 2025-09-01:
万分感谢你的帮助！